vsdip/IndiaRiscvTapeoutProgram #17: [OTHER]  Gate level Simulation

nomika-ch, 2025-09-27:
### Week

week-0

### Describe Your Issue

When I am trying gate level simulation, i am not getting the GLS simulation output i.e., even though i am doing GLS i am getting rtl waveform.

dhanvantibhavsar, 2025-09-27:
> ### Week
> week-1
> 
> ### Describe Your Issue
> When I am trying gate level simulation, i am not getting the GLS simulation output i.e., even though i am doing GLS i am getting rtl waveform.

> ### Week
> week-1
> 
> ### Describe Your Issue
> When I am trying gate level simulation, i am not getting the GLS simulation output i.e., even though i am doing GLS i am getting rtl waveform.

First verify if you are opening a correct file also Can you explain the main difference between RTL and GLS? What does it mean when you say both RTL and gls are same or different ?Also, what should be the expected output of GLS?

nomika-ch, 2025-09-28:
Thank you, I have provided a wrong file name that was my mistake.

In RTL simulation, we check the behavior of the verilog code and timing is not considered, for gate level simulation the netlist which is generated after synthesis is simulated which even checks for timing(which includes the real gate and interconnect delays mentioned in SDF). So both RTL simulation and GLS simulation are not same. I was expecting _1_,_2_ related waveforms.

closed 2025-09-28 by nomika-ch