vsdip/IndiaRiscvTapeoutProgram #19: [DOUBT]

Ketan-009, 2025-09-27:
### Week

week-2

### Your Question

Sir,

I noticed that the theory and lab references(hyperlinks) provided in the PDF are appearing only as plain text. They do not redirect to any link or Project Labs when clicked.

Could you please check this issue?

ankitmawle, 2025-09-27:
Thank you for letting this be know, will get it updated asap

KAMATHAM19, 2025-09-28:
These are the two links

Theory - https://github.com/hemanthkumardm/SFAL-VSD-SoC-Journey/tree/main/11.%20Fundamentals%20of%20SoC%20Design

Lab - https://github.com/hemanthkumardm/SFAL-VSD-SoC-Journey/tree/main/12.%20VSDBabySoC%20Project

ankitmawle, 2025-09-28:
Issue resolved on platform, links are now accessible on platform itself

closed 2025-09-28 by ankitmawle